LibreSolar/mppt-2420-lc #7: Remove CAN cable power supply

martinjaeger, 2016-07-28:
Power supply in the RJ45 jack was originally planned to power small devices like displays on the bus. However, an external short circuit would pull down the 10.4V voltage for driving the MOSFETs and could cause problems (or even destruction) in the charge controller.

Risk is too high --> remove the power supply. If the bus should still be powered, it can be done by an external device.

aShure, 2016-07-29:
How about a 100mA polyfuse in line to the 8P8C jack? Powered bus connector is a great feature for a display.

martinjaeger, 2016-07-29:
Yes, polyfuse would be possible, but increases part count and cost. Currently I'm not even sure how many people will use the CAN bus feature at all. So a dedicated external device to power the bus for those people who really use the bus could be a better idea.

martinjaeger, 2016-12-17:
Closed with updated 20A capable board revision.

closed 2016-12-17 by martinjaeger