LibreSolar/mppt-2420-lc #14: Questions about hardware

therealchalz, 2017-05-07:
Not an issue per se, but I had a few questions about your hardware design, just for curiosity's sake.
* Most similar MPPT designs I've seen use a mosfet to prevent power going into the panels at night.  However in most of the other designs I've looked at, the mosfet would be on the positive solar side with the gate tied to the high-side buck mosfet.  On your design the mosfet is on the negative solar side and tied to the low side mosfet.  I can see how both would work, but I was just wondering if it made a difference and if there was a specific reason you did it this way.
* I see that your current shunt resistors and opamps are on the negatives of the circuits (for the battery current and DC loads, for example).  I can see one advantage would be not be having to deal with common-mode voltages on the current shunts; is that why you did it this way, or are there other reasons?

Both of these questions just come from my personal preference of always having things grounded (like the negatives for the solar leads or the DC loads/battery always being grounded) and having controls and current shunts on the high lines, but I'm curious about what I might be missing.

Nice work!

martinjaeger, 2017-05-12:
Really good questions. You are addressing two aspects on which I deliberated a lot.

1. In a [previous version](https://github.com/LibreSolar/MPPT-Charger_20A/blob/0a19fcb8f357c8d85dbe0a4e154765cdb30ef952/MPPT_Charger_12A.pdf) I had the reverse blocking MOSFET on the high side, exactly as you described. However, I had some issues with the reverse polarity protection, as the pull down resistor goes to the "outside" of the circuit (i.e. solar panel). If you now apply a reverse polarity, you have to make sure that the reverse voltage cannot be "seen" by the gate of the buck converter HS gate and (worse) the gate drive IC. So I designed this other circuit with the PMOS and the diodes between the two gates (see link above). That circuit worked well in my LTspice simulation, but not in reality. After a few seconds of reverse polarity, the MOSFET started conducting and destroyed the circuit.
With the MOSFET placed on the low side, the reverse polarity protection works out of the box without any additional effort. As you only connect the solar panel to this port, I don't see any issues with the ground level offset. But in general I agree that it is nicer to have no broken ground connection anywhere.

Remark: The reverse polarity protection only works if panel voltage + battery voltage is less than the max Vgs = 60V of the MOSFET. If you connected a 24V battery and afterwards connect a 40V solar panel, the protection might not work anymore. For a 12V system it should always be OK. To be safe, the battery should be connected after the solar panel.
Maybe I should choose a 100V rated MOSFET for the reverse blocking to be safe in all cases

2. The sense resistors at the low side were mainly chosen to be able to use cheap OpAmps for current sensing. Unfortunately, I realized that the LM358 does not go down to 0V as much as I expected. This makes the current measurement at very low currents quite inaccurate which might cause problems in energy counting if you have low power loads over a long period of time. That's why I used the DAC of the STM32 to generate a bias voltage for the differential amplifier and be able to measure currents in positive and negative direction. Currently this was only implemented for the DCDC current measurement, which makes it possible to use it as a Boost converter aswell. In the next revision I will also use this approach for the load current to be more accurate. Unfortunately I see a quite high noise on the DAC output which I have to investigate further...

therealchalz, 2017-05-18:
Very interesting, excellent.  Thanks for taking the time to respond.  Also good tip about the lower voltage limit on the reverse polarity protection; that was nonobvious to me.
Keep up the good work!

shaileshbakshi, 2017-11-24:
I want code of mcu used please send it to me my email id is shaileshbakshi77@gmail.com

martinjaeger, 2017-12-01:
@shaileshbakshi what do you mean? The firmware? It can be found [here](https://github.com/LibreSolar/MPPT-Charger_Software).

closed 2018-02-05 by martinjaeger