LibreSolar/mppt-2420-lc #18: how do you protect the ADC input from reverse polarity of the battery?

kikiloaw, 2018-01-08:
i am trying to figure out your circuit. the output voltage of voltage divider will go higher because of reverse polarity.

martinjaeger, 2018-01-08:
The output reverse polarity is achieved with the fuse and the body diode of the LS MOSFET Q4. This should happen fast enough so that the large output capacitor can hold down the voltage until the fuse is blown. Otherwise also the capacitor would be damaged.
The reverse polarity protection does not work if using a low-power current-limited device, such as SMPS, instead of a battery.

kikiloaw, 2018-01-08:
so it means doesn't need to add polarity protect at the battery side?

kikiloaw, 2018-01-08:
can this build protect from reverse polarity at adc and capacitor without fuse? i cant find high current fuse here. i am trying to build at 60->100 Amps

martinjaeger, 2018-02-05:
I think for high current applications you need to think about another way of reverse polarity protection. Maybe MOSFETs?

closed 2018-02-05 by martinjaeger

kikiloaw, 2018-02-08:
hi, i am having a problem with my mppt. i try to calibrate each of my
voltage divide separetely. but then when the it starts charging. the
voltage at the battery side will detect arround 20 volts. i dont know that
wrong with that

martinjaeger, 2018-02-08:
Did you rebuild the Libre Solar MPPT? Which revision?

Please describe your problem in more detail.

kikiloaw, 2018-02-08:
I build my own version using ir2104 since it is the driver I already
have.,  I use your schematic as reference on how you build and how you code
them all ., can please take a look at mine? Ican send you my schematic

kikiloaw, 2018-02-08:
I build my own version using ir2104 since