LibreSolar/mppt-2420-lc #19: reverse polarity circuit

fhlb-tech, 2018-02-05:
Hi 
Nice work mate
Just Wondering, why didn't you use the basic reverse battery protection circuit similar to the schematic below?
https://i.stack.imgur.com/a4mNC.png

[image]

Are you switching Q2 with the same PWM signal that's regulating the output voltage/current!? That will probably add ripple on the DCDC_in node and require you to add more capacitors to lower the ripple there.

martinjaeger, 2018-02-05:
The typical reverse polarity protection is not suitable here. Q2 has two function: Reverse polarity and reverse current blocking from battery into the PV panel. With your above circuit there will still some leakage current also at very low solar input voltage, so the battery can be depleted over night.

The switching of Q2 via the LS gate driver is not an issue, as it stays on once the PWM is running. The pull-down current to switch it off is blocked by the diode. It is only switched of by the pull-down resistor after the PWM completely stopped.

fhlb-tech, 2018-02-05:
Thank you for your explanation. your right

closed 2018-02-05 by fhlb-tech

komuch, 2020-05-18:
> The switching of Q2 via the LS gate driver is not an issue, as it stays on once the PWM is running. The pull-down current to switch it off is blocked by the diode. It is only switched of by the pull-down resistor after the PWM completely stopped.

Correct me if i am wrong:

1. Using only upper mosfet (and adding diode) for async mode is not possible in this case.
2. The only option is to run PWM in complemetary mode for a few cycles and disable lower mosfet.

I am still learning and I still have to learn a lot :)

martinjaeger, 2020-05-18:
1. Correct.
2. Probably this won't work either, as the gate capacitance will discharge again after a few cycles.

Why do you want to use discontinuous mode?