LibreSolar/mppt-2420-lc #23: PWM frequency selection

fhlb-tech, 2018-02-14:
Hi
Can you explain why did you choose a switching frequency of 70kHz and not higher?

martinjaeger, 2018-02-14:
If you further increase the frequency, the efficiency goes down, especially at low load. I found 70 kHz as a good compromise.

AdiXen, 2018-03-15:
> Can you explain why did you choose a switching frequency of 70kHz and not higher?

I have a bit of experience with MPPT controller design.  Actually lowering frequency within reason is more efficient in most cases.    I settled on 64khz for my designs, which in my case was built using an 8 bit PIC running at 64Mhz which also allowed for a 1000 step duty cycle (0-100%).   I was initially experimenting with 100khz to 128khz and the efficiency was dropping noticeably over a lot of the output range.   The main advantage with higher frequency is smaller inductor and capacitors but you get more switching core loss in the inductor, higher mosfet driver power consumption and higher mosfet switching losses and in most cases this loss is over the whole load range so at lower outputs your efficiency suffers a lot. 
If tiny and low cost is your goal then up the frequency, and deal with the trade off in efficiency.
If efficiency above all else is what you are going for and don't care about size or cost of components, go the lowest frequency practical, but i'd advise staying above human hearing 22khz, or it will annoy :)  i'd say something like 32khz-80khz and up is probably optimal.

martinjaeger, 2018-03-15:
I did not go below 70 kHz, but I did an efficiency measurement for 70 kHz and 100 kHz:

[image]

The inductor is 22 uH, so I think it does not make sense to go below 70 kHz. An inductor with same size and 33 uH has almost twice the DC resistance which has to be considered aswell.

martinjaeger, 2018-03-15:
@AdiXen Are your MPPT designs open source? If yes, I would appreciate to have a look... I'm always curious to see which design choices other people took in order to rethink and improve my own ideas :)

AdiXen, 2018-03-15:
22uH is fine for your voltage range and frequency.   Well depending on what your toroid core permeability is, as I don't actually know :)

Do you use the InductorRing calculation program?  It is by far the best calculator to determine things like core loss in various input/output ranges and core saturation point .   

I wouldn't say my MPPT designs are open source, but neither do I ever intend to sell them so happy to talk about them.  I don't really have a pretty or complete schematic or even pictures online.  I do have a schematic in Diptrace but it's hideous 👎

AdiXen, 2018-03-15:
The top one is my most latest design.     Granted it's not pretty to look at - only about 40% of it is SMD components, but 99% could be replaced if I wanted to go to the effort of updating the PCB and buy all the replacements for components I already have in through hole, these designs were never meant for mass production :)

The LCD or OLED display (interchangable 16 x 2) is not pictured but mounts on the right.

[image]

martinjaeger, 2018-03-18:
Thanks for posting your pictures. I didn't know the InductorRing tool and did calcultions currently only manually (Excel-based). I am using off-the-shelf inductors from Feryster, who offer inductors with a broad range of inductance and current.

atochukwu0, 2018-06-30:
On my own experience, I used 80kHz, but my mppt is based on 3 phase or dual phase asynchronous,  I tested my mppt with input 80v 12a and getting output 54v 17.2a.

atochukwu0, 2018-06-30:
https://scontent.flos5-1.fna.fbcdn.net/v/t1.0-9/fr/cp0/e15/q65/36240675_2095851514020367_4374224217298698240_o.jpg?_nc_cat=0&efg=eyJpIjoidCJ9&oh=d6fc07e18a6522993a02098b0a15273d&oe=5BE8C568

atochukwu0, 2018-06-30:
https://user-images.githubusercontent.com/37390469/37467112-81b30afa-28b3-11e8-9826-d02b36fa5a47.jpg

atochukwu0, 2018-06-30:
[image]

atochukwu0, 2018-06-30:
About 97.8% efficiency 😎, using 100uh inductor, I2c lcd or EaDog lcd serial. Maximum system current 50a using two inductor  70 a using 3 inductors, 120degree switching. Pic18f46k22, I used internal master clear to avoid external noice resetting the system, 80kHz is very good since I have been using my mppt, very stable. Even at very high current, no system malfunction.

AdiXen, 2018-07-02:
Thanks for posting those pictures atochukwu

I have managed to get my single phase synchronous design to peak at 99.3% efficiency in a 48v system, for example with 64v in to 52v out when pushing 600 watts.  I have just incrementally improved that design and hoping for at least 99.4% peak, possibly 99.5% under very ideal circumstances, like 60v > 54v.
   
I can also get > 99% efficiency in a 48v system anywhere from 5 amps out up to full output (50A).  Obviously the efficiency drops a bit if you use lower voltage battery, at 24v it only peaks over 99% at around 9 amps output up to 25 amps output, and in 12 volt mode it only peaks at 98.8%.
I was interested in trying to create a multiphase design and still am, but I am just not sure I can invest the massive amount of time required to 'beat' my current design.  Obviously multiphase enjoys other benefits like smaller inductors and capacitors. but i'm all about the efficiency above all else.

How did you get the out of phase PWM output on a PIC18F46K22 working? difficult?

What sort of tracking algorithm are you using and how stable is it?

That picture you posted of your MPPT kind of looks a bit like the MakeSkyBlue commerical MPPT:

[image]

atochukwu0, 2018-07-02:
How did you get the out of phase PWM output on a PIC18F46K22 working? difficult? 

Definitely, is not difficult, after ups and down, I used timer 2 for phase 1, timer 4 for phase 2 and timer 6 for phase 3 , during pwm initialization, I start the timers 120 degree apart for two 3 phase or 180 degree for dual phase, but first the pwm out pin will first set as input to read the states of the mosfet gate driver, if 3 drivers present it will set the timers 120 degree else 180 degree.

atochukwu0, 2018-07-02:
What sort of tracking algorithm are you using and how stable is it? I used simple and efficient tracking method Hill climbing, but implemented with state machine, very efficient, the phase version of my mppt goes up to 98.8% ,but dual phase is 97.8% ,because I used asynchronous.

atochukwu0, 2018-07-02:
That picture you posted of your MPPT kind of looks a bit like the MakeSkyBlue commerical MPPT: 

Yes, my mppt development was inspired by make sky. I bought one from China, wanted to buy another one, but due to import duty from clearing goods from China in Nigeria, I was not able to buy, so I started the development, clearly studying make sky, they used 80kHz, so I settled at 80khz instead.

atochukwu0, 2018-07-02:
[image]

martinjaeger, 2018-07-02:
Wow, very interesting. Are you also going to publish your design on github? Would be very interested to get to know more details of your design.

AdiXen, 2018-07-02:
Very cool, love to see your design in action, I'm not a huge fan of the MakeSkyBlue MPPT though, I got one and tested it out, It's not great, has a number of issues, tracking is poor, and the readout accuracy is so poor that my 50A rated unit shows 50amps output when its really only outputting 40A, and also limits itself to that as it thinks it really is outputting its maximum 50A.  It could possibly be that my unit is faulty though.

atochukwu0, 2018-07-03:
AdiXen , I will post some videos soon, the problem of outputting 40a while displaying 50a can possibly be fault from current shunt resistor heated up ( if you used shunt)  , heat will make it loose it's resistance and act as a load, try increase try the current sensor.

AdiXen, 2018-07-03:
I tried replacing the flimsy wire shunts with a quality bourns surface mount type rated for far higher currents,  <a href="https://au.mouser.com/ProductDetail/Bourns/CSS2H-5930K-1L00F?qs=sGAEpiMZZMtlleCFQhR%2fze%252bwwzj9NwmYhDgjZqz1AaRzsMrgtuWgEA%3d%3d">Here</a>, hits the same issue!  
  poor filtering before input to microcontrolller or not enough averaging in software I suspect!

atochukwu0, 2018-07-03:
There is a 5 pin chip, possibly could be the linear amplifier used to sense the shunt, you can trace the wiring around the sensing amplifier (5pin chip)  , lower the gain of the amplifier , but the readings won't be accurate, am sure the out current will increase, Co's lowering the sensing amplifier will trick the controller chip thereby allowing more current to flow, instead of getting stucked at 40a.

atochukwu0, 2018-07-03:
I once had the problem, so I hat to lower the gain of the linear amplifier I used, I was using 68 gain before and 2 Milli ohm , but now I use 1 Milli ohm and 31 gain, the current senses up to 50a no issues.

atochukwu0, 2018-07-04:
[image]

The mppt packaged. 😎

atochukwu0, 2018-07-29:
Adixen mail me, atochukwu0@gmail. Com...