LibreSolar/mppt-2420-lc #24: Pull Down Schottyk Diodes

AdiXen, 2018-03-15:
Hi,

Just wondering if you have considered adding Schottky diodes in parallel with R1 & R3 3.3 ohm gate resistors?

At a guess you currently have somewhere around 100ns-150ns switching times, for both on and off?
Adding schottky diodes should improve the switching off time and boost the efficiency of the regulator slightly without any drawbacks, no added switch node peaks/ringing or other detrimental effects, though possibly switching deadtime might need be adjusted in software slightly (less deadtime)

You could still add these diodes without making a change to your board by stacking these on top of your existing resistors, albeit I realise that's not perfectly neat.

Other than that, I see you already have a lightweight snubber in place and have almost perfect layout of the input bypass caps to mosfet arrangement so switch node peak and ringing is minimised as much as possible, and your inductor core is already sendust (not sure of the permeability rating though, 60ish is probably optimal)  Those mosfets you are running about the best you can get in this voltage range for this application.  I can't see why you couldn't hit 99% *peak* efficiency when using a 24 volt battery bank and panels are operating at a MPP of around 30-32 volt at around the 10 amp output mark, which is more or less as good as it gets.  Have you done efficiency testing yet?

martinjaeger, 2018-03-15:
Hi, thanks for your valuable feedback!

According to my calculation, the MOSFET driver is already at its limit of 1.4 A pull-down current with the 3.3 Ohm gate resistors, so I would probably overload it with a Shottky diode in parallel. In order to reduce ringing and still have fast switch-off behavior, I added the bootstrap resistor which slows down only the switch-on of the high side MOSFET.
Do you still think that diodes could bring a huge efficiency improvement? I could try to add some for testing (as suggested by you). But for the final design a more expensive driver with higher current would be necessary in that case.

Regarding efficiency: The dead-time with 300 ns is currently not yet at its limit. With the previous revision of the board (power stage very similar) I reached efficiencies of 98.7% peak (at around 7 amps). So your guess was very good.

With hand-soldered boards I measured voltage ringing of up to 40V at 31V input voltage and 14V/10A output without snubber. With the snubber, the ringing was reduced to 35V, which is still not perfect, as it increases further at 20A.

Now I got some boards with proper automatic soldering and I was positively surprised. The ringing was more or less gone. Didn't expect the process to have so much influence. Maybe the snubber resistor can be reduced a bit to gain more efficiency.

According to your experience: Which maximum temperature would you allow for the MOSFETs. I measured surface temperatures at the MOSFETs of around 100°C (thermal camera) with 50% duty cycle and 20A output at 14V. The on-board thermocouple close to the MOSFETs measures slightly lower temperature. I would probably start power derating at 80°C measured temperature. What do you think?

AdiXen, 2018-03-15:
Schottky diodes in parallel with the gate resistors is fairly normal.  Using 3.3 ohm gate resistance is a fairly good trade off because if is helping to prevent ringing (spikes) on the switch node on the on cycle, ideally you would have less resistance but there is always a trade off.  Actually if you look at the LM5107 datasheet, there is a diagram for typical application, in that diagram I can see pull down diodes in parallel with the gate resistors, if it's in the datasheet i'm sure that means they recommend it 👍 
I am using PMEG10010ELR diodes for instance, but this is not very critical which schottky diode you use, as long as it's 1 amp+ rated.  Overall it presents a fairly acceptable load to the Mosfet driver as it's only an extra load during the off cycle, and the Mosfet gate itself is just a capacitor and not an unlimited power source that it has to fight against 100% of the time.

I'm not going to lie adding these diodes will not make a huge difference to efficiency, especially not when you are running quite low gate capacitance mosfets, as the 3.3 ohm resistors will still allow your mosfets to run at around 150ns off time or thereabouts I would guess anyway.   But it adds such little cost and has so few drawbacks its definitely worth trying.

As far as deadtime goes, you want some overlap between the on and off time of the high and low mosfet for optimum efficiency.  You generally don't need to worry about shoot through as much as you would think, again within reason.  I am for instance am using 60ns deadtime, and in my MPPT design, that is optimum for me.  As your Mosfets in your design probably turn on a bit slower than my design and it looks like they have lower gate thresholds you probably want 100-200ns deadtime for the most optimum configuration, have you tried experimenting in this regard?  Again don't take my word for it, i'm not sure what your timing is like without seeing an oscilloscope probe on your gates, I just did some quick maths based on my own experiences and some of the components you have used :)

Personally I am using the FAN7191_F085 gate driver, which is 4.5 amp peak, but I am also using 150v Mosfets with twice the gate charge, and the entire mosfet stage is paralleled (6 mosfets) so if you look at it such that I have 4 times the gate capacitance to drive than in your design the mosfet drivers actually have similar overall loads.

The board layout of the power stage is all critical, heaped solder and that kind of thing increases parasitic inductances = increased spikes.   To me your layout looks nearly perfect, the small mlc bypass capacitors on the input being as close as humanly possible to the source of the high side mosfet and drain of the low side mosfet is the most critical part.  Some ringing is of course just unavoidable especially as voltages and amps increase.

The snubber resistor does not really control the power usage of the snubber, again within reason, the capacitor value does mainly. Usually the resistor of 1-2 ohm is normally the right value in most designs.   Your snubber cap is quite small (1n) and wouldn't be wasting much power, something like 0.16 watts @ 50 volts input and much less as the input voltage drops, i.e 0.04watts @ 25v input. I would leave it if your peak switch node voltage values are at all concerning (when measured drain to source on a decent oscilloscope) i.e potentially going too far over your Mosfet VDS rated value and pushing it into avalanche territory when running higher input voltages and amperage loads.  

If you have breathing room in regards to the switch node ringing discussed above, to gain the most efficiency I would reduce your bootstrap resistor value to like 1 ohm along with adding the pulldown diodes, which will speed up both your on and off times of both synchronous mosfets and then add a reasonable reduction in deadtime in software.  Overall it's probably enough to get you that 99% peak mark though, with these changes.

The trade off I have made in my latest design, beyond optimal mosfet and input bypass capacitor placement, was to use 150V mosfets for a 100v max input design so I can allow the switch node ringing to shoot up to 150V (and a little bit higher) safely so I don't have to make many compromises in the snubber or gate resistors. , I am using 1 ohm resistors on the high side gates and 4.7 ohm on the low side gates, no bootstrap at all, I instead use self contained 1w isolated dc-dc converters to power the high side so that I can have 100% on time If I choose to, so that I can have optimal power extraction at all power outputs and MPP voltages.

All this said i'm not a professional electrical engineer.  I am an 'advanced' hobbyist that just has a real interest in MPPT and DC converter designs and I am up to MPPT Prototype number 14 of my own, which I have been working on off and on for over a decade now and I think they are getting quite good by now.  I'm mostly focused on underlying efficiency and extracting every last watt both in dc conversion and tracking than pretty bells and whistles, that said I do have some bells and whistles in my designs anyway :) 
What i'm trying to say is, i'm giving you food for thought, don't take everything i say as gospel, your design looks very good all on it's own.

On your last point, my latest two designs do have ntc thermistors near the mosfet power stage to detect over heating, and I have it set to 80 degrees disable, and re-enable at 70 degrees.  My MPPTs are passively cooled designs also, but I've never had it kick in because it's very efficient.  I'm running one design permanently on an 1100 watt 42v Open circuit solar power array into a 24 volt battery bank so it's really operating in an optimum efficiency most of the time (either above or near 99%), and 60 degrees C is probably the highest I have ever seen on a 30 degree ambient day.