LibreSolar/mppt-2420-lc #25: Noise radiating copper planes

Mike4U, 2018-03-15:
Studying  about SMPS in various books but especially 'Switching Power Supplies A-Z', Sanjaya Matiktala, they suggest that circuit traces with high start and stop currents should not be spread out on planes.  While short and fat as possible, there are diminishing returns in inductance when width is even close to length.  So in spots like below the 'Libre Solar MPPT' text that copper is not helping inductance but is making a big antenna for radiating noise.  Also signals like this should be on one layer with ground on the opposite side for a two sided design.

Looks like your effectively using top and bottom planes for thermal as well.  I don't know how to have GND on the bottom in that location and still have the thermal vias.  However, as long as your using thermal vias they will make a huge difference thermally if they are directly under the part as shown in the data sheet.  Even better if they are pre-filled with solder.

Just my 2 cents and I'm still learning so be interested in your thoughts.

AdiXen, 2018-03-16:
This is a good read in regards to noise:

[https://www.richtek.com/m/~/media/AN%20PDF/AN045_EN.pdf](url)

Generally though, a large ground plane spanning an entire side is a good thing for the most part.

martinjaeger, 2018-03-16:
Right, the mentioned plane below the Libre Solar MPPT text is necessary to spread the heat of the HS MOSFET. It's always a compromise between thermal requirements and electrical requirements. The most critical section for noise radiation is the path of high current between input capacitor and HS drain to LS source. This section is kept as short and close together as possible.

Reagarding thermal vias I read that the vias below the pads can cause issues if they suck the solder and you have too little solder left to connect the part properly. So I decided to put the vias right next to the part instead of below the pad. It seems to be effective enough, as the bottom layer temperatures reach around 90°C when having 100°C on top of the MOSFET housings. As you also mentioned, getting a thermally connected ground plane below the input voltage plane is not easily possible.

I will do some EMC tests end of April and hope to get some more insights here.

Mike4U, 2018-03-16:
I was mistakenly looking at the CSD18543QSA data sheet for the 12A part that shows the thermal vias.  Don't understand why the CSD18540Q5B 29A data sheet doesn't mention them.
Also, (100-20)°C*1W/50C= 1.6 W dissipation, on just this part.  Don't see how the efficiency mentioned on other issue can be close to 98% unless the RθJA is much worse than advertised.

AdiXen, 2018-03-16:
In my own experience I currently purposely use 4 x 1.5mm evenly placed holes under surface mount TO263-7 mosfets.  Now in no way am I saying this is optimal, however I have found the partial flow of solder paste down into the holes during reflow (hot air) forms a well set solid filled hole that seems to be an advantage rather than a loss.   I have no issues with electrical resistance, and the solder running into the holes improves the heat transfer quite a bit to the other side of the board.  Whether doing more smaller holes might still have this affect I don't know but I would say the smaller the hole the less chance the solder is going to drain as much.  I understand that technically more smaller holes has a better thermal resistance on paper (more plated copper) however i'm not so sure in practice when you take into account the solder.   That said, only 10 degrees difference between package and ground plane is good when running at those temperatures so probably not much of an issue here anyway.

Mike4U, 2018-03-16:
AdiXen, your link above didn't work but found it here:
[http://www.richtek.com/Design%20Support/Technical%20Document/AN045](url)
Great article.

Mike4U, 2018-03-17:
Sorry about closing and I can't seem to get my link to work either.

AdiXen, 2018-03-17:
> AdiXen, your link above didn't work but found it here:

Sorry I have also updated my link now using old school html programming instead of the link button, works now :)

I have re-read that article a number of times, always get something out of it.   It's teachings applies to both large and powerful and small synchronous buck converters just the same.

martinjaeger, 2018-03-18:
> Also, (100-20)°C*1W/50C= 1.6 W dissipation, on just this part. Don't see how the efficiency mentioned on other issue can be close to 98% unless the RθJA is much worse than advertised.

The stated efficiency was measured at 10A output current (my Multimeter couldn't handle more current), but it stays far below 100°C in this operating point.

100°C are reached at 20A output current (14.4 V) and it fits quite well my expectations. My calculation says approx. 1.6 W conduction and switching losses in the HS FET, slightly less in the LS FET, almost 3 W in the inductor and 6 W in total, which corresponds to 98% efficiency at approx. 300 W.
The size of the copper areas is less than the datasheet setup used to measure RthJA. In addition to that, I have several heat sources very close. So I don't expect it to reach the advertised 50 K/W under free convection conditions without heat sink.

Mike4U, 2018-11-02:
Sounds reasonable, 
thanks

closed 2018-11-02 by Mike4U