LibreSolar/mppt-2420-lc #26: Development Timeline And Commercial Availailability

vanarchist, 2018-03-17:
Hello,

I know the project is in active development, but was wondering if you had a schedule for when it would be production ready? Also, are you planning to manufacture and sell this? I am interested in using it for my solar setup. I do have experience assembling PCBs, just don't have any of my gear with me now, could probably hit up a hackerspace though.

Thanks

martinjaeger, 2018-03-18:
I am planning to perform some EMC tests and finalize the design this summer, probably within a Kickstarter campaign. Currently, you can only order the PCBs and parts yourself and solder them in your local hackerspace.

vanarchist, 2018-03-18:
Have you heard of [Utopian](https://join.utopian.io/)? It's a project running on the STEEM blockchain to support open source projects. You might be able to get some funding through that for development. I can't help much hardware wise with my current setup, but I could help with documentation, part sourcing, and other stuff.