LibreSolar/mppt-2420-lc #27: high power dissipation in 25A fuse

fhlb-tech, 2018-04-04:
Hi
I was just doing some tests on ATO fuses (from little fuse). 20A going through 20A fuse gave me an 80 degree temperature rise with 85mV drop (1.7W of heat). 20A going through 30A fuse gave me 44mV drop. I'm just saying that this fuse will get real hot (and any components near that fuse will get hot too) so take note.

zmaragdus, 2019-05-23:
Has anyone considered throwing in an ideal diode controller instead of a fuse? This would cut down the power dissipation and also eliminate the issue of having to replace a fuse. A good ideal diode controller costs $1-$3, a beefy MOSFET another $0.50-$1, and some misc. passive components.

martinjaeger, 2019-05-24:
Where would you suggest to place the ideal diode?

The fuse is useful for reverse polarity protection and overcurrent protection at the same time. And current can flow in both directions (buck mode or boost mode of the DCDC) depending on the application. I guess this would be difficult to achieve with an ideal diode. What do you think?

zmaragdus, 2019-06-29:
I was thinking of putting it in the place of F1 (DC/DC Power Stage, page 2 of the schematics). I suppose you're right, though, in that this would break the design for someone who wanted to flip it around and use it as a boost. I don't know of an ideal diode controller that is easily "reversed," either.

Regarding the overcurrent thing, you could go with a larger / more complex surge stopper (back-to-back MOSFETs plus current sense resistor), but then we're back at the problem of needing to easily reverse the connections for someone who wants to flip the design.