LibreSolar/mppt-2420-lc #28: Reproduction of PCB via OSE CNC-Mill

mikelovic, 2018-07-07:
idea of [OSE online meeting from 03 july 2018](https://www.youtube.com/watch?v=OUrtiV383zg&t=2455s):
- reproduction of MPPT PCB with [OSE CNC Mill](https://wiki.opensourceecology.org/wiki/D3D_CNC_Circuit_Mill)
- adaption of circuit layout to requirements of CNC mill
- possiblities, ideas, opinions, ... ?!

case06, 2018-07-08:
This is probably not the ideal pcb-project for reproduction with circuit-mill, because its mainly consisting of smd-parts instead of throughole-parts and contains a huge amount of vias, which are used here even for thermal conduction too.

So any "adaption" does probably mean doing a complete new layout and could probably only done by Shane, who is the only one who knows about the special requirements for its cnc-circuit-mill.

Also the smd-soldering may be a problem, so one theoretically must do a complete re-engeneering of the schematics with replacing as many smd-parts with through-hole parts as possible.  Maybe the remaining bigger smd-chips can be more modularized and become provided as breakout-boards.

In total, i am under the impression that the idea to reproduce this pcb with the circuit-mill was mainly caused by the whish to make some practical use of the mill, but this pcb is not well suited for the approach. So its maybe easier and saves tons of wasted work and menpower, if we produce the complete circuit in the regular way and send one piece over to them.

rkrenzler, 2018-07-08:
I see vias a the main issue here. Is it worth to invest humanpower in the solution of the vias-Problem?
https://hackaday.com/2015/02/25/diy-through-hole-plating-like-a-boss/

Can we automate it?