LibreSolar/mppt-2420-lc #29: low side mosfet

kikiloaw, 2018-07-31:
hi, i made my own MPPT using ir2104 driver. I kept burning my lowside mosfet. before the current flow backward through the mosfet, the lowside mosfet must be off. how did you manage that to prevent that from burning?

martinjaeger, 2018-07-31:
You need enough dead-time between switching of the different MOSFETs. Check your MPPT with an oscilloscope and you will see if this is the reason.

martinjaeger, 2018-08-01:
That document is very interesting, thanks for sharing.

Actually, I don't switch off the LS MOSFET during reverse current, so the converter stays in continuous current mode (CCM). Reverse current should not be the reason for your failure. However, switching to discontinuous mode (DCM) might increase the efficiency a bit. See also here: www.ti.com/lit/an/slyt358/slyt358.pdf

To allow DCM in synchronous buck converters, you need to generate the PWM signal separately for LS and HS, so your IR2104 with fixed dead-time is not feasible. In addition to that, you would need a fast current measurement in the inductor path with high sample rate. Or you estimate the time when the current goes negative based on a model.

Did you check your switching behavior with an oscilloscope? Maybe your gate resistors are too large so that one MOSFET is not yet completely turned of when the other is switched on. But this is all hard to judge because I don't know your schematic.

AdiXen, 2018-08-01:
With that particular driver chip it would be very important to operate the shutdown pin before or at the exact time as you stop the PWM output for whatever reason, otherwise the low side mosfet would go high which is basically a dead short across the battery when on for any amount of time.

There is also a high chance parasitic spikes are killing your low side mosfet.  These can easily add up to be multiple times your input voltage.    The biggest thing you can do to reduce this is make sure you have your input capacitor as close as humanly possible to the drain of the high side mosfet - and the source of the low side mosfet.   I'm talking every millimeter counts in this regard.   Best to have a lower value mlcc capacitor as physically close as possible followed by a electrolytic capacitor(s) a little bit further away but still as close as possible.
You can easily blow 150v rated mosfets with just 64v on the input and 24v output for instance if you do not have your input capacitors close enough, i've done it, multiple times in early prototypes.

kikiloaw, 2018-08-01:
yes it is a very interesting documentation about buck converting being used as a charger.
https://github.com/kikiloaw/mppt
i uploaded my schematic and pcb which is made in eagle cad.
i am using only throughole package since smd is very rare to find here in our place. 
i change the mosfet driver to mic4102 so that when my problem is not yet solve. i can operate in non synchronous mode only. and if the problem is fixed. i can directly enable the driver for  synchronous mode. hope you can help me. thank you very much

martinjaeger, 2018-08-01:
I had a look at your board layout. The HS MOSFET is very far away from the input capacitors. That's exactly the problem AdiXen was talking about. The input capacitors (ideally ceramic + electrolytic) have to be placed as close as possible to HS mosfet drain and LS mosfet source.

The reverse blocking MOSFET should not be put between input capacitor and HS MOSFET, but between input capacitor and terminal.

These documents might be helpful:
http://rohmfs.rohm.com/en/products/databook/applinote/ic/power/switching_regulator/converter_pcb_layout_appli-e.pdf
https://www.richtek.com/m/~/media/AN%20PDF/AN045_EN.pdf

kikiloaw, 2018-08-01:
thank you very much, I change the order of the  reverse blocking MOSFET and the capacitor. can you please take a look if i have the right order of the components, from the input solar > current sensor > reverse mosfet > capacitor > hs mosfet.

martinjaeger, 2018-08-02:
The position in the schematic is better now, but the capacitors are still not placed well. The GND plane is very interrupted and the ground return paths from the LS source to the capacitors are very long:

[image]

In addition to that, the switch node should be as small as possible, as it has high dV/dt rates and creates EMC. It is quite large, the MOSFETs could be placed closer.

kikiloaw, 2018-08-02:
so does it mean that the high side capacitor gnd pin must be also closer to the low side mosfet source pin? the the your circled is for the drain of the high side mosfet. and the other one ground. than you very much for the biggest help.

martinjaeger, 2018-08-02:
Yes. My red lines represent the path in GND plane of high di/dt rates, as shown in the Richtek document, Figure 3. That's the most critical point.

kikiloaw, 2018-08-02:
after redesigning the PCB Layout. does it mean that burning mosfet at low side will be prevented?

martinjaeger, 2018-08-02:
I can't promise ;)

kikiloaw, 2018-08-02:
What would be other possibilities that could burn the low side mosfet. My solution is just putting diode inthe input of buck converter and the output of buck converter. But drops the efficiency very much.

AdiXen, 2018-08-03:
Think of Synchronous rectification like a modern Turbo charged engine.    Except instead of extra power from a Turbo you get extra efficiency.    Turbo charged motors if not controlled carefully can easily explode.  Same deal with that low side Mosfet, so you have to firstly tune the circuit, then tune the software to make absolutely sure it's not switching the low side mosfet on for too long or at the same time as the high side FET, the MIC4102 should prevent the latter.   Truth is the quicker and more efficiently you drive the mosfets you will have a bigger chance of blowing the thing up, so it's an engineering trade off.    It is much more simple and reliable to use plain non-synchronous conversion but as you say you lose out on efficiency, which makes more heat, so you have new things to worry about.

Another thing you should be checking is make sure in software you have safety checks about how low the duty cycle is going, you don't want it to go too low, and measure the voltage forming on the input. because synchronous rectification can convert in reverse which can also lead to failure of either Mosfet!

kikiloaw, 2018-08-03:
i choose the switching frequency of 50 KHz which give 20us period. max input of 100v and a 12v battery. then give an 12 percent duty cycle. a 12 percent duty cycle of 50 KHz is 2.4us on time. so this is the lowest duty that the pwm is set.

atochukwu0, 2018-08-15:
Rc snubber across low side mosfet can also help to eliminate emi at the low side mosfet .

atochukwu0, 2018-08-15:
Kikiloaw , At 12% duty , you are building a boost converter , not buck anymore, you are pushing current from batteries to solar, so it's likely your mosfet will blow all the time , limit your duty cycle to 40% preferably 60%  and make Max voc 50v . With these I hope your will stop turning your mosfet to magic smoke 😎..

closed 2018-10-17 by martinjaeger