LibreSolar/mppt-2420-lc #30: [Question] adding charge circuitry, when does it become economically viable?

e123h, 2018-10-17:
first off; 
@martinjaeger   If this is not the place for these types of questions, feel free discard them. 

I had read the other messages and got the idea that were people present here with a lot of experience in the field of solar systems. 

I have a question about the point where it becomes economically viable to add (more advanced) charge circuity to a solar system?

For example, considering the following (hypothetical) situation; 
PV panel:
Maximum power (Wp) :3.2W
Maximum power voltage (V): 18V
Maximum power current (A): 0.177A
Open circuit voltage (V): 21.6V
Short circuit current (A): 0.210A

Charging a lead-acid 12V, 5.2Ah battery. 

Considering the idea, that adding charge circuitry the system the total costs will be larger than the increase in performance?  In other words why not add just a couple of diodes between the PV panel and the lead-acid battery? (So the speak)

martinjaeger, 2018-10-17:
This seems to be a more generic question not related to this charge controller, but I will try to answer anyway.

Your power is very low, so you might want to use a small dedicated MPPT ASIC if you really need MPPT.  If cost is a concern, a single MOSFET might be enough. A diode is not sufficient as it does not prevent overcharging of the battery.

closed 2018-10-17 by martinjaeger

martinjaeger, 2018-10-17:
Additional remark: A diode is still needed with the MOSFET to prevent discharging the battery at night (reverse blocking diode).

e123h, 2018-10-17:
Thank you for your response.