LibreSolar/mppt-2420-lc #32: High ADC noise

7h0ma5, 2018-10-19:
I recently finished building the 20A MPPT Charger with the V0.10 PCB. It works quite well so far, but I am experiencing some problems with the voltage and current measurements. The battery voltage for example is varying by almost ±0.5V, which leads to problems with the charge mode selection.

Does anyone else experience this kind of behavior? Maybe it is just a problem with the soldering, but I checked everything thoroughly under the microscope.

To investigate the source of the noise I wrote a simple program which transfers the raw samples directly via the USB interface at a sample rate of about 100ksps. The measurements were made with only a 12V lead acid battery connected as power source.

This is the result of a 1 second measurement with 100k samples:

[image]

This is the FFT spectrum of the previous measurement. It most likely includes many aliasing effects due to the low sample rate:
[image]

This is the distribution of 1 million measurements over the time of 10 seconds:
[image]

I also took some measurements with a spectrum analyzer and the MPPT charger running the firmware https://github.com/LibreSolar/MPPT-Charger_Software/commit/978cfd34aa40a96c98e149fbff40651c077cfcf5. Again only a 12V lead acid battery was connected as power source. The lower trace (yellow) is the base level without the battery connected.

The voltage reference and the battery voltage at the voltage divider are very clean compared to the power supply of the microcontroller. The ferrite bead L4 seems to be relatively ineffective.

#### Measurement of V<sub>DDA</sub> across *C34*
[image]

#### Measurement of V<sub>DD</sub> across *C32*
[image]

#### Measurement of V<sub>REF</sub> across *D6*
[image]

#### Measurement of V<sub>BAT</sub> across *C18*
[image]

martinjaeger, 2018-10-20:
Thanks for sharing such interesting results.

All your measurements were done with the DC/DC power stage switched off, correct? So EMI created by the power stage can be excluded as a cause...

I also experienced a not perfectly stable voltage measurement, but only with a deviation of around ±0.1V (which is still a lot). With my current board an current software it seems OK.

Maybe you can check with the most recent software in the development branch where [the internal voltage reference of the STM32](https://github.com/LibreSolar/MPPT-Charger_Software/blob/d32e1a1c95ef29a3a15e6516182ca105eaba2f14/src/adc_dma.cpp#L67) is used.  In my tests the voltage reference was not very accurate (less accurate than the voltage of the LDO), so I don't really see a point of keeping it for a future revision of the board. Also the internal voltage reference is less accurate than the LDO, but I don't want to rely on the LDO voltage as it might drop in case of overload and give completely wrong results.

The new software version has an [increased ADC sample time](https://github.com/LibreSolar/MPPT-Charger_Software/blob/d32e1a1c95ef29a3a15e6516182ca105eaba2f14/src/adc_dma.cpp#L267) (necessary to read out the internal references) which might improve accuracy.

If you can share your test software, I can flash it to my board and collect some samples aswell.

martinjaeger, 2018-11-02:
Just figured out that my ADC noise gets higher if I use the SPI port. Did you have something connected to the UEXT connector during your measurements?

martinjaeger, 2019-01-05:
Found some time to further investigate this issue.

It looks like the 10V to 5V LDO (AZ1117) is not stable and causes oscillations on the supply rail, which are also transferred to the 3V3 supply rail.

Here is an oscilloscope shot of the 5V rail during idle:
[image]

The 3V3 rail looks like this:
[image]

This is probably the reason for the ADC noise, but I didn't find an easy workaround or solution yet.

All measurements were taken with the DC/DC power stage switched off and without anything connected to the UEXT port.

Dee-Fuse, 2019-07-30:
I just discovered that disabling the LED PWM (charlieplexing) reduces the ADC noise significantly.
I had similiar messurement noise with the LEDs enabled / switching, without the noise is down to a few mV. I'll messure the voltage rails and inputs with an stabilized powersupply attached later this week.

eleinur, 2019-12-12:
Hi Martin,
May be it's a stability issue on the AZ1117 regulator. The AZ1117 regulator is not stable ceramic outputs capacitors. I had the same issue on a similar design with another LDO regulator.
The AZ1117 from Diodes Inc. has been replaced by the AZ1117I. There's a small sentence on the cover page...Compatible with Low ESR Ceramic Capacitor.
The idea would be to replace the 5V regulator and check the stability.

martinjaeger, 2019-12-16:
Thanks, that's a very good observation and might actually be the reason for the noise... but the AZ1117 had a quite high quiescent current anyway, so I got rid of it completely in the new MPPT 2420 HPX design.