LibreSolar/mppt-2420-lc #33: Inductor Alternative

JuculianD, 2018-11-06:
Hello,
I am trying to rebuild your project but I can not find a suitable diode. The diode you have used in the schematic is not available to me in Germany.

Can you give me an alternative or a place to buy it?
Can I use one with slightly different inductance without changing the software?

Thank you very much,
Julian.

martinjaeger, 2018-11-07:
Hi Julian,

You can buy them in Poland at Feryster.com. But shipping cost might be quite high for one piece.

If you are living in Germany, just send me your address via email (see Libre Solar website). I still have some of them and can send one to you.

You could also use an alternative inductor with 22uH or more, but make sure it can handle >25A and high frequencies (e.g. Sendust core).

Martin

JuculianD, 2018-11-07:
What does more mean? How high can I go with the inductance?
If I can not find one I would like to get the proper one from you. 

Vielen Dank für das Angebot!

JuculianD, 2018-11-07:
Sorry for my questions but I would like to understand it more. I am a bit experienced with electronics but did not go to inductance and buck converter. I know the working principle but how does the inductance influence the output? With a higher/lower inductance, which changes have to be made and how will the output current and voltage change?
I was not able to find anything useful for me in the net.

Thank you for you explanations.

martinjaeger, 2018-11-07:
Higher inductance means less output ripple current for same switching frequency. If you increase inductance, you can lower the switching frequency (or other way round). Higher switching frequency means higher switching losses, but lower inductor size for same current. Switching frequency is currently 70 kHz and I don't recommend to go much higher.

Current rating is more or less independent, but the maximum current is also defined by other components. So even if you find an inductor for 40A, 20A output will still be the max. for the rest of the components.

So basically it's all about finding a good tradeoff between efficiency, size and cost. I don't think you will find an inductor with similar current rating and size, but much higher inductance, though.

JuculianD, 2018-11-07:
Thank you very much! I know much better now ;-).
I have found an inductor with 25A Max and 70uH. Is that okay? It is not very expensive.
Could I just lower the PWM frequency to something around 40KHz? 
Would 25A be possible with that frequency and active cooling of the mosfet?

JuculianD, 2018-11-07:
I will be under 25A anyway, but it is just for my understanding ;-). 
Obviously I would have to make thicker coppery traces etc.

martinjaeger, 2018-11-07:
It might work, but for a detailed answer some calculations would have to be done. The saturation current has to be the DC current + ripple current, so 25A might not be enough (depending on the exact spec of that inductor).

JuculianD, 2018-11-07:
Okay, that makes sense. Do you have a formula or calculator for me?

Sorry for my newbie questions but you have already helped me a lot!

JuculianD, 2018-11-07:
Another question:
you have used a LM5107 half bridge driver. It has reached its lifecycle and I can not buy them at my supplier anymore. Can I just drop in a LM5106 without any changes (100V and 1.8A Peak Output Sink current)?

martinjaeger, 2018-11-08:
Mouser has 11000 pieces of LM5107 on  stock and the TI website doesn't tell anything abo end of lifecycle. Didn't check in detail if LM5106 works as well.

JuculianD, 2018-11-09:
Very weird, I have looked at some.wrong chip version, maybe.
Anyway, I will test it and if not working, I will buy one from mousers ;-).

closed 2019-03-05 by martinjaeger