LibreSolar/mppt-2420-lc #34: Mbed Die Runtime b+

penojazij, 2019-02-14:
khgkhg
````
Archiving .pioenvs\nucleo\libabd\libAdafruit_GFX.a
Linking .pioenvs\nucleo\firmware.elf
Checking size .pioenvs\nucleo\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  43.7% (used 7160 bytes from 16384 bytes)
PROGRAM: [========  ]  76.5% (used 100228 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs\nucleo\firmware.elf
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-15:03)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 0.011031
Error: target voltage may be too low for reliable debugging
Info : stm32f0x.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x08002540 msp: 0x200011d8
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x20016448
Info : flash size = 128kbytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x200011d8
wrote 102400 bytes from file .pioenvs\nucleo\firmware.elf in 4.053876s (24.668 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x200011d8
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x200011d8
verified 100500 bytes in 0.432645s (226.848 KiB/s)
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
shutdown command invoked
`````

martinjaeger, 2019-02-14:
Did you also try the development branch of the software?

I'll have a look tomorrow. I was going push a more recent software anyway in the next days.

As this is a software related issue, it would be better to post it in the software repository.

martinjaeger, 2019-02-18:
I uploaded a completely new software last Friday. Can you please check if it works?

martinjaeger, 2019-03-05:
No response: Assuming the problem is solved.

closed 2019-03-05 by martinjaeger