LibreSolar/mppt-2420-lc #36: Stencil Pattern & Mask for the MOSFET

maclow92, 2019-03-06:
1. The Solder past stencil recommandation is 4 row 2 columns
   *According to Chapter 7.3 of Texas Instrument (SLPS488B –JUNE 2014–REVISED APRIL 2017)
2. The Mask should be updated to allow good positioning of the component during reflow
   *According to chapert 7.2 

Sorry I am not able to install the KiCad to support _ that need a
[image]
roung 5Gb :-(

martinjaeger, 2019-03-06:
Thanks for the feedback.

I took the pattern from a different 5x6 MOSFET and didn't have any issues so far. But I realized during soldering that the MOSFETs could have slightly more solder. I'll update the package according to [ON Semi AND9137/D](https://www.onsemi.com/pub/Collateral/AND9137-D.PDF) in the future to allow exchanging of the MOSFETs.

maclow92, 2019-03-06:
good idea for universal footprint

Consider also PCB pattern update : that I mix up in copy past :-/

[image]