LibreSolar/mppt-2420-lc #39: MOSFET Alternative / evolutions

maclow92, 2019-03-08:
STL135N8F7AG Could maybe match with this HW application or on new one

Pros : 80 V, 3.15 mΩ typ., 120 A.
         Automobile grade = range of T°-55 to +175°C

Cons : Gate-source voltage +/-20V with Gate threshold voltage 4.5V
          Package little bit bigger

martinjaeger, 2019-03-08:
The package looks fine, it's also 5x6 mm.

However, the maximum current will have to be decreased because of higher Rds(on). I don't think you will find another MOSFET that has the same low Rds(on) as the CSD18540Q5B, but a higher voltage rating. It's always a tradeoff between different requirements. I decided for higher current instead of higher voltage.

> Cons : Gate-source voltage +/-20V with Gate threshold voltage 4.5V

Why is this a con?

maclow92, 2019-03-11:
> However, the maximum current will have to be decreased because of higher Rds(on). I don't think you will find another MOSFET that has the same low Rds(on) as the CSD18540Q5B, but a higher voltage [...]
Yes you today deisgn is very good / just looking around to be able to connect 3 pannels of 12V in series

Higher Rds :
- Mean that an Heatsink will be needed ?
- Or it is limiting the efficiency of the product ?

> > Cons : Gate-source voltage +/-20V with Gate threshold voltage 4.5V
> 
> Why is this a con?
right no issue , today it's driven through 10V

martinjaeger, 2019-03-11:
> Higher Rds :
> 
> * Mean that an Heatsink will be needed ?
> * Or it is limiting the efficiency of the product ?

Possibly both, lower efficiency for sure.